Hyy2001X/AutoBuild-Actions-BETA #4: build new-wifi-d2 错误

lgfyxx1, 2022-12-11:
相关日志如下，谢谢大佬：
是不是aria2有啥问题？

```
Hunk #1 succeeded at 846 (offset 68 lines).
[10:30:16] Found board ramips patch file: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/CustomFiles/Patches/dualband_wifi-ramips.patch
patching file package/kernel/mac80211/files/lib/wifi/mac80211.sh
[10:30:16] Found generic patch file: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-### 
BETA/CustomFiles/Patches/fix_aria2_auto_create_download_path-generic.patch
patching file feeds/packages/net/aria2/Makefile
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored -- saving rejects to file feeds/packages/net/aria2/Makefile.rej
patching file feeds/packages/net/aria2/files/aria2.init
Error: Process completed with exit code 1.
```

xu26311365, 2022-12-14:
我也是这个问题

lgfyxx1, 2022-12-15:
> 我也是这个问题 删掉这个文件fix_aria2_auto_create_download_path-generic.patch 建议注释掉aria2,我不怎么用这个

谢谢，删掉后完美解决了:)

qazz-shyper, 2022-12-15:
@lgfyxx1 上游配置文件修改过了，导致patch匹配不到，修改fix_aria2_auto_create_download_path-generic.patch文件，删除前半部分，1~12行，修改-162,7 +162,8为-167,7 +165,8

lgfyxx1, 2022-12-15:
> 

哇，感谢感谢！ 亲测works！

lgfyxx1, 2022-12-15:
> @lgfyxx1 上游配置文件修改过了，导致patch匹配不到，修改fix_aria2_auto_create_download_path-generic.patch文件，修改7-8行
> 
> ```
> -PKG_RELEASE:=1
> +PKG_RELEASE:=2
> ```
> 
> 为
> 
> ```
> -PKG_RELEASE:=2
> +PKG_RELEASE:=3
> ```
> 
> 修改15行`-162,7 +162,8`为`-165,7 +165,8`

赞

Hyy2001X, 2022-12-18:
> @lgfyxx1 上游配置文件修改过了，导致patch匹配不到，修改fix_aria2_auto_create_download_path-generic.patch文件，修改7-8行
> 
> ```
> -PKG_RELEASE:=1
> +PKG_RELEASE:=2
> ```
> 
> 为
> 
> ```
> -PKG_RELEASE:=2
> +PKG_RELEASE:=3
> ```
> 
> 修改15行`-162,7 +162,8`为`-165,7 +165,8`

感谢大佬回复

qazz-shyper, 2022-12-19:
> 感谢大佬回复

Not a Dalao, Just an user.  我是你项目的忠实用户，嘿嘿