Hyy2001X/AutoBuild-Actions-BETA #11: NEWIFI3 D2 Action手动编译错误提示：No files matching the glob pattern found

Watson-Will, 2023-02-02:
NEWIFI3 D2 Action手动编译时
workflow到Upload Firmware to Releses出现错误提示：
  No files matching the glob pattern found.

请问是什么问题？如何解决？

Watson-Will, 2023-02-02:
0s
Run svenstaro/upload-release-action@v2
  with:
    repo_token: ***
    file: openwrt/bin/Firmware/*
    file_glob: true
    tag: AutoUpdate
    overwrite: true
  env:
    CONFIG_FILE: d-team_newifi-d2
    DEFAULT_SOURCE: coolsnowwolf/lede:master
    UPLOAD_RELEASES: true
    UPLOAD_ARTIFACTS: false
    UPLOAD_BIN_ARTIFACTS: false
    DELETE_USELESS_FILES: true
    DELETE_OLD_WORKFLOW: false
    CACHE_ACCELERATE: true
    REPO_URL: https://github.com/coolsnowwolf/lede
    REPO_BRANCH: master
    Compile_Date: 202302020358
    Display_Date: 2023/02/02
    WORK: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt
    CONFIG_TEMP: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/.config
    AutoBuild_Features: true
    x86_Full_Images: false
    AutoBuild_Fw: AutoBuild-lede-d-team_newifi-d2-R23.[1](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:1).1-[2](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:2)02[3](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:3)0202-Full-SHA256.FORMAT
    CustomFiles: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/CustomFiles
    Scripts: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/Scripts
    BASE_FILES: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/package/base-files/files
    FEEDS_LUCI: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/package/feeds/luci
    FEEDS_PKG: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/package/feeds/packages
    Default_Title: "Powered by AutoBuild-Actions"
    Regex_Skip: "packages|buildinfo|sha256sums|manifest|kernel|rootfs|factory|itb|profile|ext[4](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:4)|json"
    Version_File: package/lean/default-settings/files/zzz-default-settings
    Fw_MFormat: bin
    FEEDS_CONF: /home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/feeds.conf.default
    Author_URL: https://github.com/Watson-Will/AutoBuild-Actions-BETA
    ENV_FILE: /home/runner/work/_temp/_runner_file_commands/set_env_8affe89a-d8a[5](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:5)-4cec-83b1-af45ad2d5105
    Author: Watson-Will
    Github: https://github.com/Watson-Will/AutoBuild-Actions-BETA
    TARGET_PROFILE: d-team_newifi-d2
    TARGET_BOARD: ramips
    TARGET_SUBTARGET: mt7[6](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:6)21
    TARGET_FLAG: Full
    OP_VERSION: R23.1.1-[20](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:20)[23](https://github.com/Watson-Will/AutoBuild-Actions-BETA/actions/runs/4065132228/jobs/7012604402#step:14:23)0202
    OP_AUTHOR: coolsnowwolf
    OP_REPO: lede
    OP_BRANCH: master
    Compile_Result: true
Error: No files matching the glob pattern found.

Watson-Will, 2023-02-02:
0s
Run svenstaro/upload-release-action@v2
  with:
    repo_token: ***
    file: openwrt/bin/Firmware/*
    file_glob: true
    tag: AutoUpdate
    overwrite: true
  env:
    CONFIG_FILE: d-team_newifi-d2
    DEFAULT_SOURCE: coolsnowwolf/lede:master
    UPLOAD_RELEASES: true
    UPLOAD_ARTIFACTS: false
    UPLOAD_BIN_ARTIFACTS: false
    DELETE_USELESS_FILES: true
    DELETE_OLD_WORKFLOW: false
    CACHE_ACCELERATE: true
    REPO_URL: https://github.com/coolsnowwolf/lede
    REPO_BRANCH: master
    Compile_Date: 202302020358
    Display_Date: 2023/02/02
TARGET_PROFILE: d-team_newifi-d2
    TARGET_BOARD: ramips
    TARGET_SUBTARGET: mt7621
    TARGET_FLAG: Full
    OP_VERSION: R23.1.1-20230202
    OP_AUTHOR: coolsnowwolf
    OP_REPO: lede
    OP_BRANCH: master
    Compile_Result: true
Error: No files matching the glob pattern found.

Hyy2001X, 2023-02-14:
固件太大了，编译完超出32M了所以没生成出来

Watson-Will, 2023-02-19:
> 固件太大了，编译完超出32M了所以没生成出来

OK
不知道能不能通过修改
lede/target/linux/ramips/image/mt7621.mk
和
lede/target/linux/ramips/dts/mt7621_d-team_newifi-d2.dts
中对应的rom容量大小和分区大小来解决，就像编译64m固件那样
如果不行应该就只能精简一些软件了