Hyy2001X/AutoBuild-Actions-BETA #32: 没看明白如何实现x86_64更新后不丢失配置和插件的

Plixs, 2023-11-26:
大概看了下luci-app-autoupdate还是用openwrt中sysupgrade实现备份和更新的，但是没找到还原配置的代码，不知道如何实现的保留配置和插件的（包括手动安装的插件）
还有luci-app-autoupdate为何一定要搭配AutoBuild-Actions编译的固件使用，看了下使用AutoBuild-Actions编译openwrt时把Scripts目录下某些shell工具编译进openwrt，为何不能单独使用luci-app-autoupdate代码没找到
能否指点一下，谢谢。

Hyy2001X, 2023-11-26:
本质通过sysupgrade更新的，最多只能保留配置但是不能保留自己装的软件包，autoupdate只是提供了一个方便用户更新固件的通道。
原理是通过Autobuild-Actions中的Scripts生成相应的通道信息(如版本 更新地址 设备名到固件中，通过执行autoupdate读取这些信息然后代替用户下载固件并升级

Plixs, 2023-11-26:
我以为可以保留自己装的软件包，x86_64无损升级确实是个麻烦事，看到几个方案都特别麻烦，不如重新编译再安装。

Hyy2001X, 2023-11-26:
> 我以为可以保留自己装的软件包，x86_64无损升级确实是个麻烦事，看到几个方案都特别麻烦，不如重新编译再安装。

直接把需要的软件包编译进固件吧，方便省事

Plixs, 2023-11-26:
> > 我以为可以保留自己装的软件包，x86_64无损升级确实是个麻烦事，看到几个方案都特别麻烦，不如重新编译再安装。
> 
> 直接把需要的软件包编译进固件吧，方便省事

另外，我尝试新建了一个actions_x86.yml,修改源码分支为openwrt/openwrt:openwrt-23.05，报错找不到指定文件夹
/home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/Scripts/AutoBuild_Function.sh: line 243: cd: /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/CustomFiles/Kconfig: No such file or directory
不知道还要修改哪边

Hyy2001X, 2023-11-26:
> 另外，我尝试新建了一个actions_x86.yml,修改源码分支为openwrt/openwrt:openwrt-23.05，报错找不到指定文件夹 /home/runner/work/AutoBuild-Actions-Template/AutoBuild-Actions-Template/Scripts/AutoBuild_Function.sh: line 243: cd: [...]

看了下用的是AutoBuild-Actions-Template稳定库，那个已经很久没有去维护了，有点问题，等我晚点同步更新一下

Plixs, 2023-11-26:
> 看了下用的是AutoBuild-Actions-Template稳定库，那个已经很久没有去维护了，有点问题，等我晚点同步更新一下

谢谢回复。

closed 2023-11-26 by Plixs