Hyy2001X/AutoBuild-Actions-BETA #34: 编译出错

kwy-hu, 2023-12-06:
Run echo "Free space:"
Free space:
Filesystem      Size  Used Avail Use% Mounted on
/dev/root        84G   [38](https://github.com/kwy-hu/AutoBuild-Actions/actions/runs/7100761095/job/19327507526#step:8:39)G   46G  45% /
devtmpfs        7.9G     0  7.9G   0% /dev
tmpfs           7.9G  4.0K  7.9G   1% /dev/shm
tmpfs           1.6G  1.1M  1.6G   1% /run
tmpfs           5.0M     0  5.0M   0% /run/lock
tmpfs           7.9G     0  7.9G   0% /sys/fs/cgroup
/dev/sda15      105M  6.1M   99M   6% /boot/efi
/dev/sdb1        63G  4.1G   56G   7% /mnt
tmpfs           1.6G     0  1.6G   0% /run/user/1001

#
# configuration written to .config
#
[20:38:[43](https://github.com/kwy-hu/AutoBuild-Actions/actions/runs/7100761095/job/19327507526#step:8:44)] [Firmware_Diy_Before] Starting ...
[20:38:43] Entering directory: /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt ...
### VARIABLE LIST ###
WORK=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt
CONFIG_TEMP=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/.config
AutoBuild_Features=true
x86_Full_Images=false
AutoBuild_Fw=AutoBuild-lede-jdcloud_re-sp-01b-R23.11.11-20231205-Full-SHA256.FORMAT
CustomFiles=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/CustomFiles
Scripts=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/Scripts
BASE_FILES=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/base-files/files
FEEDS_LUCI=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/feeds/luci
FEEDS_PKG=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/feeds/packages
Default_Title="Powered by AutoBuild-Actions"
Regex_Skip="packages|buildinfo|sha256sums|manifest|kernel|rootfs|factory|itb|profile|ext4|json"
Version_File=package/lean/default-settings/files/zzz-default-settings
Fw_MFormat=bin
FEEDS_CONF=/home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/feeds.conf.default
Author_URL=https://github.com/kwy-hu/AutoBuild-Actions
ENV_FILE=/home/runner/work/_temp/_runner_file_commands/set_env_12431657-c0e0-4e51-8f00-c849ba201ea9

[20:38:43] [Firmware_Diy_Before] Done
[20:38:43] [Firmware_Diy_Main] Starting ...
[20:38:43] Entering directory: /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt ...
[20:38:43] Checking out package [AutoBuild-Packages] to package/other ...
[20:38:[44](https://github.com/kwy-hu/AutoBuild-Actions/actions/runs/7100761095/job/19327507526#step:8:45)] Done
[20:38:44] [C] Copying /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/CustomFiles/Depends/tools to /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/base-files/files/bin ...
[20:38:44] Done
[20:38:44] [C] Copying /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/CustomFiles/Depends/profile to /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/base-files/files/etc ...
[20:38:44] Done
[20:38:44] [C] Copying /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/CustomFiles/Depends/base-files-essential to /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/base-files/files/lib/upgrade/keep.d ...
[20:38:44] Done
[20:38:44] [C] Copying /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/CustomFiles/Depends/coremark.sh to feeds/packages/utils/coremark ...
[20:38:44] Done
[20:38:44] [C] Copying /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/CustomFiles/Depends/banner to /home/runner/work/AutoBuild-Actions/AutoBuild-Actions/openwrt/package/base-files/files/etc ...
[20:38:44] Done
[20:38:44] [Firmware_Diy_Main] Done
[20:38:44] Checking out package [luci-theme-argon] to package/themes ...
[20:38:[45](https://github.com/kwy-hu/AutoBuild-Actions/actions/runs/7100761095/job/19327507526#step:8:46)] Done
[20:38:45] Checking out package [luci-app-openclash] to package/other ...
Error: Process completed with exit code 1.

Hyy2001X, 2023-12-06:
https://github.com/kwy-hu/AutoBuild-Actions/blob/a87376fd43bd4de057e60db2a3bf4f94a8220051/Scripts/AutoBuild_DiyScript.sh#L70

这行删除就可以了
或者替换
AddPackage git other OpenClash vernesong dev

kwy-hu, 2023-12-06:
好，唔该先。

closed 2023-12-13 by Hyy2001X