Hyy2001X/AutoBuild-Actions-BETA #36: 编译成功后无法收集固件。求助

ilxp, 2024-01-08:
https://github.com/ilxp/AutoBuild-Actions-BETA/actions/runs/7440593440/job/20241760241。提示：

Run cd openwrt
[08:11:30] [Firmware_Diy_End] Starting ...
[08:11:30] [08:11:30] Actions Avaliable: 13G
### FIRMWARE OUTPUT ###
x86

[08:11:30] [Firmware_Diy_End] Done

没有像大佬编译的：
Run cd openwrt
[01:12:[56](https://github.com/Hyy2001X/AutoBuild-Actions-BETA/actions/runs/7387716557/job/20097014631#step:11:57)] [Firmware_Diy_End] Starting ...
[01:12:56] [01:12:56] Actions Avaliable: 23G
### FIRMWARE OUTPUT ###
config.buildinfo
feeds.buildinfo
openwrt-x86-64-generic-kernel.bin
openwrt-x86-64-generic-squashfs-combined-efi.img.gz
openwrt-x86-64-generic-squashfs-combined.img.gz
openwrt-x86-64-generic-squashfs-rootfs.img.gz
openwrt-x86-64-generic.manifest
packages
profiles.json
sha256sums
version.buildinfo

[01:12:56] Moving [openwrt-x86-64-generic-squashfs-combined-efi.img.gz] to [AutoBuild-lede-x86_64-R23.11.11-20240102-UEFI-Default-5[60](https://github.com/Hyy2001X/AutoBuild-Actions-BETA/actions/runs/7387716557/job/20097014631#step:11:61)00.img.gz] ...
[01:12:56] Moving [openwrt-x86-[64](https://github.com/Hyy2001X/AutoBuild-Actions-BETA/actions/runs/7387716557/job/20097014631#step:11:65)-generic-squashfs-combined.img.gz] to [AutoBuild-lede-x86_64-R23.11.11-20240102-BIOS-Default-d8216.img.gz] ...
/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt
[01:12:56] [Firmware_Diy_End] Done

试了，多次，找不到原因，麻烦帮忙看看。

Hyy2001X, 2024-01-08:
已pr，试试

ilxp, 2024-01-08:
谢谢，我去试试

ilxp, 2024-01-08:
大佬，问题依旧。https://github.com/ilxp/AutoBuild-Actions-BETA/actions/runs/7443605069/job/20248708449。

[16:48:59] [Firmware_Diy_End] Starting ...
[16:48:59] [16:48:59] Actions Avaliable: 13G
/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/Scripts/AutoBuild_Function-oprx.sh: line 377: cd: "/home/runner/work/AutoBuild-Actions-BETA/AutoBuild-Actions-BETA/openwrt/bin/targets/x86/64": No such file or directory
Error: Process completed with exit code 1.

再帮忙看看。

closed 2024-01-10 by ilxp