Hyy2001X/AutoBuild-Actions-BETA #37: 新三路由器 openwrt Clash版2024.1.5版本  管理无法保存设置

Soupure, 2024-01-15:
https://www.right.com.cn/forum/thread-4047888-1-1.html 下载的clash版本，管理界面设置无法保存，重启后直接丢失

[image]

Soupure, 2024-01-15:
Mon Jan 15 17:54:54 2024 cron.err crond[14682]: crond (busybox 1.36.0) started, log level 8
Mon Jan 15 17:55:38 2024 daemon.err uhttpd[3447]: touch: /etc/crontabs/root: Read-only file system
Mon Jan 15 17:55:38 2024 daemon.err uhttpd[3447]: sed: can't create temp file '/etc/crontabs/rootXXXXXX': Read-only file system
Mon Jan 15 17:55:38 2024 daemon.err uhttpd[3447]: crontab: can't create /etc/crontabs/root.new
Mon Jan 15 17:55:38 2024 daemon.err uhttpd[3447]: crontab: can't append to /etc/crontabs/cron.update
Mon Jan 15 17:55:53 2024 daemon.info hostapd: wlan1: STA 86:70:64:85:45:58 IEEE 802.11: authenticated
Mon Jan 15 17:55:53 2024 daemon.info hostapd: wlan1: STA 86:70:64:85:45:58 IEEE 802.11: associated (aid 1)
Mon Jan 15 17:55:53 2024 daemon.notice hostapd: wlan1: AP-STA-CONNECTED 86:70:64:85:45:58 auth_alg=open
Mon Jan 15 17:55:53 2024 daemon.info hostapd: wlan1: STA 86:70:64:85:45:58 RADIUS: starting accounting session CD2B510DD479B4AD
Mon Jan 15 17:55:53 2024 daemon.info hostapd: wlan1: STA 86:70:64:85:45:58 WPA: pairwise key handshake completed (RSN)
Mon Jan 15 17:55:53 2024 daemon.notice hostapd: wlan1: EAPOL-4WAY-HS-COMPLETED 86:70:64:85:45:58
Mon Jan 15 17:56:35 2024 daemon.err uhttpd[3447]: touch: /etc/crontabs/root: Read-only file system
Mon Jan 15 17:56:35 2024 daemon.err uhttpd[3447]: sed: can't create temp file '/etc/crontabs/rootXXXXXX': Read-only file system
Mon Jan 15 17:56:35 2024 daemon.err uhttpd[3447]: crontab: can't create /etc/crontabs/root.new
Mon Jan 15 17:56:35 2024 daemon.err uhttpd[3447]: crontab: can't append to /etc/crontabs/cron.update
Mon Jan 15 17:57:03 2024 daemon.err uhttpd[3447]: touch: /etc/crontabs/root: Read-only file system
Mon Jan 15 17:57:03 2024 daemon.err uhttpd[3447]: sed: can't create temp file '/etc/crontabs/rootXXXXXX': Read-only file system
Mon Jan 15 17:57:03 2024 daemon.err uhttpd[3447]: crontab: can't create /etc/crontabs/root.new
Mon Jan 15 17:57:03 2024 daemon.err uhttpd[3447]: crontab: can't append to /etc/crontabs/cron.update
Mon Jan 15 17:57:29 2024 daemon.err uhttpd[3447]: touch: /etc/crontabs/root: Read-only file system
Mon Jan 15 17:57:29 2024 daemon.err uhttpd[3447]: sed: can't create temp file '/etc/crontabs/rootXXXXXX': Read-only file system
Mon Jan 15 17:57:29 2024 daemon.err uhttpd[3447]: crontab: can't create /etc/crontabs/root.new
Mon Jan 15 17:57:29 2024 daemon.err uhttpd[3447]: crontab: can't append to /etc/crontabs/cron.update
Mon Jan 15 18:02:24 2024 cron.err crond[16940]: crond (busybox 1.36.0) started, log level 8
Mon Jan 15 18:02:38 2024 cron.err crond[17792]: crond (busybox 1.36.0) started, log level 8

Hyy2001X, 2024-01-15:
空间塞满了

closed 2024-01-15 by Soupure