Hyy2001X/AutoBuild-Actions-BETA #50: 手动在线升级处于网络连接错误

ilxp, 2024-10-26:
大佬，通过tools里面的u和自动升级，都可自动选择镜像加速。但是手动升级，检测完版本后，只要点击升级更新，一直报错网络错误。路由器已经翻墙。tools里网络检测，goolge链接也是正常的。
[image]

Hyy2001X, 2024-10-26:
试一下`autoupdate --verbose`
这个日志更加详细

ilxp, 2024-10-26:
luci界面升级时提示的，采用tools进行升级会自动进行代理。luci界面手动升级就会提示网络错误。

Hyy2001X, 2024-10-26:
> luci界面升级时提示的，采用tools进行升级会自动进行代理。luci界面手动升级就会提示网络错误。 在手动更新--检测更新，获取到新版本后，点击执行更新。立马跳转到更新日志里提示网络错误。 ![1729946707961](https://private-user-images.githubusercontent.com/61609258/380423534-5d45e0d0-cfa7-47cc-8246-174081531783.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5NTc2OTQsIm5iZiI6MTcyOTk1NzM5NCwicGF0aCI6Ii82MTYwOTI1OC8zODA0MjM1MzQtNWQ0NWUwZDAtY2ZhNy00N2NjLTgyNDYtMTc0MDgxNTMxNzgzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI2VDE1NDMxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTNjMTUwY2FhODU1N2RmOWI0ZDdmZmQxOTYxM2E2YTQwMmFhZWU4NGU2MzIzOWU2OTJhNWUxODMyNmEyNDc0ZDYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.mxwG9vgjk7n8uuaSepv0RxRXWlQ-A2MqN8zQDvP89GE) ![1729946733607](https://private-user-images.githubusercontent.com/61609258/380423536-fedd2fa1-f079-4d3e-ba8e-9b60ad9de881.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3Mjk5NTc2OTQsIm5iZiI6MTcyOTk1NzM5NCwicGF0aCI6Ii82MTYwOTI1OC8zODA0MjM1MzYtZmVkZDJmYTEtZjA3OS00ZDNlLWJhOGUtOWI2MGFkOWRlODgxLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMjYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDI2VDE1NDMxNFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3NjkzZWU2NDE4OTY4YjZhOTI3MTYxZjQwOWJlYWExNWQ3MzYwMjNlNDhiMTkxMGVhYmRhNzZkYTU2NjM5YzMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.0-t6KraKHYC0hvh-gnhqByLjBASZIA9YqJNVkDZzdqY)

手动更新--检测更新 执行的是`autoupdate -u`命令，所以请手动运行一下`autoupdate -u --verbose`命令获取更加详细的日志信息，谢谢！

ilxp, 2024-10-27:
Openwrt-AutoUpdate Script by Hyy2001 V6.9.12
ps: unrecognized option: e
BusyBox v1.36.1 (2024-09-23 12:34:46 UTC) multi-call binary.

Usage: ps

Show list of processes

        w       Wide output

[21:27:21] 特殊指令: [定时更新] [详细信息] | /sbin/sysupgrade -q

[21:27:21] 执行: 更新固件

[21:27:25] **Google 连接错误,优先使用镜像加速下载!**

[21:27:25] 正在检查固件版本更新 ...
--2024-10-27 21:27:26--  https://api.github.com/repos/ilxp/oprx-builder/releases/latest
Resolving api.github.com... 20.205.243.168
Connecting to api.github.com|20.205.243.168|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 13577 (13K) [application/json]
Saving to: '/tmp/autoupdate/API_Cache'

/tmp/autoupdate/API_Cache                  100%[========================================================================================>]  13.26K  --.-KB/s    in 0.09s   

2024-10-27 21:27:27 (143 KB/s) - '/tmp/autoupdate/API_Cache' saved [13577/13577]

ilxp, 2024-10-27:
执行 autoupdate -u --verbose 后，显示  Google 连接错误,优先使用镜像加速下载! 然后启用了镜像加速。这个更新命令换为autoupdate -u -P  即可

ilxp, 2024-10-27:
同时，有两个bug：1、tools工具： 3. 更新固件 [保留配置]   应该是 不保留配置。2、  autoupdate里面的代理，大佬你的那个挂了，https://ghproxy.com代理变成了https://ghp.ci

ilxp, 2024-10-27:
建议添加这3个代理：https://cdn.jsdelivr.net/gh   https://raw.gitmirror.com  https://ghp.ci

Hyy2001X, 2024-10-27:
已修改为`ghp.ci`，并加入`hub.gitmirror.com` `gh.con.sh`镜像源作为备选

Hyy2001X, 2024-10-27:
> 同时，有两个bug：1、tools工具： 3. 更新固件 [保留配置] 应该是 不保留配置。2、 autoupdate里面的代理，大佬你的那个挂了，[https://ghproxy.com代理变成了https://ghp.ci](https://ghproxy.com%E4%BB%A3%E7%90%86%E5%8F%98%E6%88%90%E4%BA%86https://ghp.ci)

已完成修改，谢谢

Hyy2001X, 2024-10-27:
> luci界面升级时提示的，采用tools进行升级会自动进行代理。luci界面手动升级就会提示网络错误。 在手动更新--检测更新，获取到新版本后，点击执行更新。立马跳转到更新日志里提示网络错误。 [...]

最新脚本将网络检测的地址改为了`www.baidu.com`，请再尝试一下看看

closed 2024-10-30 by ilxp