Hyy2001X/AutoBuild-Actions-BETA #52: luci-app-autoupdate有个bug

ilxp, 2024-10-31:
大佬，最新版6.10.1使用下来没有问题，非常好。但是luci-app-autoupdate这个还是有个bug没有得到解决。就是在js的luci上，点击执行更新，还是跳转到日志里，没有执行，显示网络连接错误，请稍后再试。空了看下。
https://private-user-images.githubusercontent.com/61609258/381160233-1370366a-e3ae-4ff3-bd3a-0b9c9add301d.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzAzNjA3NDMsIm5iZiI6MTczMDM2MDQ0MywicGF0aCI6Ii82MTYwOTI1OC8zODExNjAyMzMtMTM3MDM2NmEtZTNhZS00ZmYzLWJkM2EtMGI5YzlhZGQzMDFkLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDEwMzElMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMDMxVDA3NDA0M1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTY3YTI1MWU1YWY2OWYwN2QzMTVlNDljOGUyODVkZWQ1N2RlNzBlMjE3MDRhZmE2ZWQ0NmQ1ZDQzN2MwODFhYmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.o3DOANzl2CmLi4aMMZAGIN3pOX079lR3SjZpuBK0gF0

ilxp, 2024-10-31:
[image]

Hyy2001X, 2024-10-31:
执行一下`ping www.baidu.com -c 1 -W 3`截图我看看

ilxp, 2024-10-31:
[image]
这个功能在lean大的lede的luci18上没有问题，回去执行升级。但在js的luci上就会出现网络连接的问题。

ilxp, 2024-10-31:
现在我刷的固件是lean大佬的ledeluci1806的，点击执行更新，会跳转到日志界面：
[21:58:27] [27845] 执行: 更新固件
[21:58:31] [27845] Google 连接检查结果: [000]
[21:58:31] [27845] Google 连接错误,优先使用镜像加速下载!
[21:58:31] [27845] 正在检查固件版本更新 ...
[21:58:31] [27845] 删除文件: [/tmp/autoupdate/API]
[21:58:31] [27845] 检查软件包: [wget-ssl] ... 错误
[21:58:32] [27845] 删除文件: [/tmp/autoupdate/API_Cache]
[21:58:32] [27845] 执行下载指令: [curl --silent --insecure -L -k --connect-timeout 5 --retry 1 -o /tmp/autoupdate/API_Cache https://api.github.com/repos/ilxp/oprx-builder/releases/latest]
[21:58:34] [27845] 开始解析 Github API ...
这里就是正常的。【实际上我已经翻墙了，谷歌是连通的】

但在js的luci上点击执行更新，就会显示网络连接错误。

Hyy2001X, 2024-10-31:
> 现在我刷的固件是lean大佬的ledeluci1806的，点击执行更新，会跳转到日志界面： [21:58:27] [27845] 执行: 更新固件 [21:58:31] [27845] Google 连接检查结果: [000] [21:58:31] [27845] Google 连接错误,优先使用镜像加速下载! [21:58:31] [27845] 正在检查固件版本更新 ... [...]

js luci版本固件用命令行autoupdate -u是可以正常更新的吗

ilxp, 2024-10-31:
可以

ilxp, 2024-10-31:
在命令行执行都没有问题，不是autoupdate的问题，是luci-app-autoupdate的问题

Hyy2001X, 2024-10-31:
> 在命令行执行都没有问题，不是autoupdate的问题，是luci-app-autoupdate的问题

应该是lean js luci的锅，先前测试了官方或者imm的23.05js luci都是正常的

ilxp, 2024-10-31:
我就是官方的23.05js luci不正常，lede的最新jsluci没有使用

ilxp, 2024-10-31:
你空了再测试下，我这边测试两个都不行，2305和master的luci都不行啊。

Hyy2001X, 2024-10-31:
> 你空了再测试下，我这边测试两个都不行，2305和master的luci都不行啊。

我这里没有js的测试环境，需要你把https://github.com/Hyy2001X/AutoBuild-Packages/blob/c66340fbf9d0e287a954bf5006243afe0264ad90/autoupdate/files/bin/autoupdate#L648
到645行删除再试一下

ilxp, 2024-10-31:
好的。

ilxp, 2024-11-01:
已经测试，就是这个648-654这段网络检测，去掉就OK了。但是又出现个问题，使用autoupdate -u 他会进行版本的判断。而luci界面的这个更新，直接就更新了，没有进行版本的对比。本地版本号和云端的一样，他还是，二话木说，直接开始更新了。

Hyy2001X, 2024-11-01:
> 已经测试，就是这个648-654这段网络检测，去掉就OK了。但是又出现个问题，使用autoupdate -u 他会进行版本的判断。而luci界面的这个更新，直接就更新了，没有进行版本的对比。本地版本号和云端的一样，他还是，二话木说，直接开始更新了。

脚本更新了一版，需要你协助点一下luci界面的更新，然后截图一下日志

ilxp, 2024-11-01:
在还有2个小建议：
1、启动方式UEFI与BIOS。现在efi已经兼容bios了，不需要编译出两个启动方式的固件格式了。只需要一个efi的即可。那么问题来了。就不需要检索上游固件的UEFI或BIOS了。

2、固件格式问题。目前大佬取的.img.gz.这个固件格式有待商榷。最好的是ext4或者squashfs。

综上2个问题：我是这样实现的的：
1、在固件命名的时候加一个字段：Fext4或者Fsquashfs。自己编译肯定知道是ext4还是squashfs。
AutoBuild-lede-x86_64-R24.9.18-20241015-Fsquashfs-UEFI-Full-c347f.img.gz

2、然后在autoupdate的第569行：
eval format=$(egrep -o '\.[a-z]+.+' <<< ${name} | cut -c2-10) 
改为：eval format=$(egrep -o '\F[a-z]+' <<< ${name} | cut -c2-9)   
取到的ext4还是squashfs。（多个F这个好取到）。
3、最后在603行Info=前面加一个字段：
FW_Format=F$(df -Th | grep "^/dev/root" | awk '{print $2}')  
#这个是取本地的文件系统格式，加个F是和命名统一。
4、将569行的info=这个字段中的${x86_Boot_Method} 改为 ${FW_Format}  不在将启动方式作为检索固件的标签。将${OP_REPO}、${TARGET_PROFILE}、${FW_Format}、${TARGET_FLAG}这4个元素作为检索的标准。

我已经实现，上游提供了ext4或者squashfs格式的固件。我就能根据自己的系统来检索固件。
[image]
[image]

ilxp, 2024-11-01:
改后的10.1新本没有问题：点击过去会更新，也会进行版本检测了。10.2还没有试。
[image]

closed 2024-11-01 by ilxp