Hyy2001X/AutoBuild-Actions-BETA #54: autoupdate v10.2无法更新

ilxp, 2024-11-01:
大佬 ，你调整了134行的for i in wget curl uclient-fetch。wget没有反应，还是的for i in curl uclient-fetch wget  优先用curl。我系统wget和curl都有安装。

Hyy2001X, 2024-11-01:
请将134行改为	`for i in ${DL_DEPENDS[@]}`试一试

ilxp, 2024-11-01:
还有 这个地址 https://www.google.com/generate_204    网络不通

ilxp, 2024-11-01:
没有反应
[image]

Hyy2001X, 2024-11-01:
[image]
不会吧，这个地址是各种代理软件用于检测连通性的呀，我这里开了代理以后访问OK的

ilxp, 2024-11-01:
我代理业开了。能上google，但autoupdate还是现实google没有联通

ilxp, 2024-11-01:
134行还得改回去：for i in curl uclient-fetch wget   这个才行

ilxp, 2024-11-01:
[image]

ilxp, 2024-11-01:
[image]
[image]

Hyy2001X, 2024-11-01:
> 我代理业开了。能上google，但autoupdate还是现实google没有联通

可能是curl wget版本差异问题，我检查一下

ilxp, 2024-11-01:
我系统是23.05.5：
wget-ssl 1.21.4-1
curl | 8.7.1-r1

ilxp, 2024-11-01:
在576 578 多了2个参数，感觉没有用处啊，反而多了很多固件信息。
LOGGER "$Info"和LOGGER "$Result"

[21:58:40] [29210] 环境变量: [Author=ilxp]
[21:58:40] [29210] 环境变量: [Github=https://github.com/ilxp/oprx-builder]
[21:58:40] [29210] 环境变量: [TARGET_PROFILE=x86_64]
[21:58:40] [29210] 环境变量: [TARGET_FLAG=oR]
[21:58:40] [29210] 环境变量: [OP_VERSION=R23.05.5-20241102]
[21:58:40] [29210] 环境变量: [OP_AUTHOR=openwrt]
[21:58:40] [29210] 环境变量: [OP_BRANCH=openwrt-23.05]
[21:58:40] [29210] 环境变量: [OP_REPO=openwrt]
[22:03:01] [30167] 结束其他 autoupdate 进程 ...
[22:03:01] [30167] 运行: /bin/autoupdate -u
[22:03:01] [30167] 特殊指令: [定时更新] | /sbin/sysupgrade -q
[22:03:01] [30167] 执行: 更新固件
[22:03:05] [30167] [curl] [https://www.google.com/generate_204] 连通性测试失败!
[22:03:05] [30167] Google 连接错误,优先使用镜像加速下载!
[22:03:05] [30167] 正在检查固件版本更新 ...
[22:03:05] [30167] 删除文件: [/tmp/autoupdate/API]
[22:03:06] [30167] 执行下载指令: [wget-ssl --quiet --no-check-certificate --tries 1 --timeout 5 -O /tmp/autoupdate/API_Cache https://api.github.com/repos/ilxp/oprx-builder/releases/latest]
[22:03:07] [30167] 开始解析 Github API ...
[22:03:07] [30167] 可用固件/日志: OprX-openwrt-x86_64-oP-R24.11.1-20241101-Fsquashfs-uefi-06d07.img.gz
[22:03:08] [30167] 可用固件/日志: OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 可用固件/日志: OprX-openwrt-x86_64-oR-R23.05.5-20241103-Fext4-uefi-2031c.img.gz
[22:03:08] [30167] Github API 解析成功!
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 固件名称: OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] squashfs
[22:03:08] [30167] 固件格式: squashfs
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 4
[22:03:08] [30167] 下载次数: 4
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 2031c
[22:03:08] [30167] 校验信息: 2031c
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] R23.05.5-20241102
[22:03:08] [30167] 固件版本: R23.05.5-20241102
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 20241101115059
[22:03:08] [30167] 发布日期: 20241101115059
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 130.36MB
[22:03:08] [30167] 固件体积: 130.36MB
[22:03:08] [30167] OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz        squashfs        4     2031c    R23.05.5-20241102    20241101115059 130.36MB        https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] 固件链接: https://github.com/ilxp/oprx-builder/releases/download/firmware/OprX-openwrt-x86_64-oR-R23.05.5-20241102-Fsquashfs-uefi-2031c.img.gz
[22:03:08] [30167] [All] 可用运行内存: [1712M]
[22:03:08] [30167] [/tmp] 可用存储空间: [940M]
[22:03:08] [30167] 未检测到已部署的云端日志,跳过下载 ...
[22:03:08] [30167] 当前固件 [R23.05.5-20241102] 已是最新版本,无需更新!

Hyy2001X, 2024-11-01:
> 在576 578 多了2个参数，感觉没有用处啊，反而多了很多固件信息。 LOGGER "$Info"和LOGGER "$Result"
>
> [21:58:40] [29210] 环境变量: [Author=ilxp] [21:58:40] [29210] 环境变量: [Github=https://github.com/ilxp/oprx-builder] [21:58:40] [29210] [...]

这个是测试的时候用的，忘记删掉了

Hyy2001X, 2024-11-01:
> 我系统是23.05.5： wget-ssl 1.21.4-1 curl | 8.7.1-r1

更新了一版本V6.10.3，加了一点测试参数，请再测试一下
可以加参数手动指定`-D curl` `-D wget` `-D wget-ssl`

ilxp, 2024-11-01:
好的，稍等

ilxp, 2024-11-01:
ok，没有问题了。

closed 2024-11-01 by ilxp